pioneers/PiElectrical #47: PDB Rev 9B erratta

dHutchings, 2016-11-18:
This is the list of PDB rev 9B erratta as currently discovered by Doug.  There may be more out there:
Erratta:
1. Wrong diameter holes (#10-32, should be #8-32) are used on the PCB for the DC converter
2. Footprint for the circuit breaker has issues:
-the footprint is symmetric, and it shouldn't be because the circuit breaker itself isn't
-the bottom hole needs more exposed copper to make soldering / bolting easier.
3. The Buzzer pin isn't a PWM on the arduino, so the buzzer can't actually buz.  We need to swap the "Calib" and "Buzzer_Signal" lines on the arduino.
4. Really need better silkscreen for the buttons, and high precision resistors.  I pushed the wrong button several times.

Improvements:

1.  The arduino currently is directly hooked up to the 5V line.  There should be a fuse added.
2.  We can delete SJ1.  It's unneeded now.
3.  The calibration Button is to be replaced with a Jumper.  That way, a student won't be able to accidentally trigger it but PiE Staff can still quickly re-calibrate it.
4. mounting holes are weirdly located.  More importantly, it seems like there will be a * small * upper enclosure area around the battery buzzer circuits, and there currently aren't enough mounting points there.  Check with enclosure people as to the dimensions of the holes - they may need to be #4-40's.
5. Need mounting for COTS battery buzzer just in case.

dHutchings, 2016-11-18:
Addendum to improvements: The 5-V anderson (red on left) is backwards relative to the 12V rail.  The 12V rail is in compliance with our long-standing standards (AKA the grizzly).

There's talk of replacing the 5V andersons with some other kind of connector, idk, ask someone about that.  Either way, replace the connector, or fix the orientation.

dHutchings, 2016-11-18:
Need to add a 3.3k (ish?  maybe 10k) resistor between 5V and GND, to help with turn-off performance.

tobinsarah, 2016-11-22:
[] Wrong diameter holes (#10-32, should be #8-32) are used on the PCB for the DC converter
Footprint for the circuit breaker has issues:
[] the footprint is symmetric, and it shouldn't be because the circuit breaker itself isn't
[] the bottom hole needs more exposed copper to make soldering / bolting easier.
The Buzzer pin isn't a PWM on the arduino, so the buzzer can't actually buz. We need to swap the
[] "Calib" and "Buzzer_Signal" lines on the arduino.
[] Really need better silkscreen for the buttons, and high precision resistors. I pushed the wrong button several times.
Improvements:

[] The arduino currently is directly hooked up to the 5V line. There should be a fuse added.
[] We can delete SJ1. It's unneeded now.
[x] The calibration Button is to be replaced with a Jumper. That way, a student won't be able to accidentally trigger it but PiE Staff can still quickly re-calibrate it.
[] mounting holes are weirdly located. More importantly, it seems like there will be a * small * upper [] enclosure area around the battery buzzer circuits, and there currently aren't enough mounting points there. Check with enclosure people as to the dimensions of the holes - they may need to be #4-40's.
[] Need mounting for COTS battery buzzer just in case.

dHutchings, 2016-11-22:
Updated list:

Schematic:
-Did not add Buzzer-disabling jumper.  (Nor changed the resistor values)
-Almost every resistor on the schematic is missing its value label.
-Wrong part for APM fuse.  The fuse part is currently wrong in the schematic (it's a jumper.  It should be a fuse)
-Rename "Calibration Button" to "Calibration Jumper"
-This PDB design is using anderson connectors for the 5V rail.  According to Seiya, these were supposed to be switched to ATX connectors.  He'll know which polariy convention to follow.
-Didn't add mounting for COTS battery buzzer.

Routing
-Use Round Vias, not square ones
-Don't need trage for GND for JP2.  Should work fine without that trace, because ground planes.
-Location of R17 isn't logical.  It'd make sense to keep it somewhere near the Power Regulator, not hanging out in the middle of the Battery Buzzer.
-Looks like circuit breaker is still symmetrical.  However, the real circuit breaker isn't.

Silkscreen:
It looks much better, but could still use some work:

-7-segment label is proportional, not vector font.  Could also be bigger.
-Same for "LED 2" label
-Could use a label next to JP2 telling what it is
-Draw a box around R2 thru R7.  Then label the box "High Precision"
-The silkscreen on the fuses is offset.  On some fuses, the label is to the left.  On others, it's to the right.

closed 2016-12-08 by dHutchings