pioneers/PiElectrical #63: Twist it fixes for yogibear

dHutchings, 2017-02-08:
Running list of fixes for yogibear for next year:

Enable falls from 4.65 (VCC) to 4.5 V when entering an error state.  The amount the voltage falls is set by a resistive network on the motor driver chip, i.e. something that we can't change.  However, this change in voltage is readable from an ADC:
- [ ] Connect the "EN" pin to an ADC - capable Arduino Pro Micro pin.

dHutchings, 2017-02-16:
- [ ] Change the pull-up value for the encoders to 1k.

dHutchings, 2017-02-27:
- [ ] Add the pull-ups for the encoders back.  Value should be as stated above.
- [ ] The form factor for the yogibear's red board leaves something to be desired.  Unique board-to-board connection strategies (AKA two right-angle .1" headers) may allow a flatter package.

onibrow, 2017-08-19:
Slide over the red board connector so the andersons can be soldered in straight up

closed 2017-10-15 by onibrow