pioneers/PiElectrical #76: Pull-up Resistors on TwistIt are screwing with the line followers

onibrow, 2017-04-27:
Make the pull up network before the resistors, aka a 2x3 resistor network. 

[image]

Move the box on right to before the resistor.

onibrow, 2017-06-06:
This would entail making a 2x3 jumper part, sourcing it, and replacing it on the twistit board.

dHutchings, 2017-06-07:
The pull-ups aren't needed at all for the line followers.

AKA: If we correctly produced the Twist-its (If we knew when to DNP the resistors), we would not have had pull-up problems..

onibrow, 2017-06-07:
I think this is slightly wrong. We weren't using the pull ups for the line followers, as they were powered and produced their own value. The issue comes because the pull up network is interfacing at a node that **all three** read pins are semi connected to. The readings become poor when one of the sensors reads a high, causing a pull up to the other readings as their is no drain, effectively pulling up the entire network. The fix would be have the jumper connect the three terminals to the five volts, not have the a collection of the  terminals connect to the 5V which was the solution this year.

dHutchings, 2017-06-10:
Or, we can just DNP R1, R2, and R3.  That will fix the problem with the line followers (and was how we fixed it last year)

The original intent was that R1...3 would either be populated or not, depending on the needs of the specific sensor attached to the board.  This was not followed in production, and the original cause of the line follower's response tracking each other.

[image]

onibrow, 2017-06-11:
Ooooh i see thanks for that, I like that solution better.

closed 2017-09-21 by onibrow